romainrossi/make_your_robot_tick #1: Sample question about the rules for the organisation comitee

romainrossi, 2018-06-11:
Here is a sample question to ask to the organisation team about the rules.
- Don't forget to add the "question" tag if your issue is a question
- Use the "bug" tag to report an error in the documents
- Choose a meaningful title, be precise and complete on your requests
- Before answering a question, always search if a similar question is not already asked
- Always be respectful in your questions or comments
- Cite the extract of the rulebook you refer to (example below)
> MYRT_RULES_V1.2 page 15
> 3.4.3 Emergency Button
>  For security reasons, all robots, must have an emergency button (in red) that is easily accessible (on top of the robot) to the referee. The referee can press this button in case of a major issue and thereby stop the match. If the referee presses this button the robot must come to a halt instantly.
> The organising committee reserves the right to reject any system it deems dangerous. 

# Answers

The MYRT organization team will give official answer to questions in batches about every weeks.
If enough modifications are made to the rules, a new version of the document will be published and will supersede the previous version.

Welcome on board and good luck !

romainrossi, 2018-07-06:
Test notifications

closed 2018-07-17 by romainrossi

SachinGoud-24, 2018-09-03:
#question
Could you kindly provide details on the voltage levels of the charging stations? That would help us a lot in designing the storage system.

romainrossi, 2018-09-04:
Please, create your questions as new "Issues" and not as comment on existing one (especially a closed one).
Moreover, this question is already answered in another issue #4

SachinGoud-24, 2018-09-04:
Sorry for that, i shall ask my other questions as you have directed. Thank
you very much.

Regards,
S Sachin Goud