romainrossi/make_your_robot_tick #3: Power Sources and Competition Platform

me17b016, 2018-07-16:
In the pdf,it is not mentioned what will the voltage of power sources and also, please clarify the difference between the different types of power souces because in page 4 in the pdf,it is written that there will be three types of power sources -regular ,accelerated and fast but In page no.5,Graph of slow,medium and fast is given with their respective activation and deactivation timing,this is really confusing... and also please tell us about nature of energy storage station(ESS),that means, will the potential between the Terminals of ESS will increase with charging or not.
In competition platform,there is no white line towards fast power sources as we have in other types of power sources and also ,distance between energy storage station and it's very near white line is not mentioned,
And please tell us why you have given that the voltage between any two terminal of our robot must not go above 48 volts,why only 48 volts,why not 50 or any other number?

romainrossi, 2018-07-17:
Your question contain multiple items which I will answer separately for increased clarity.
I will link here the related issues.

romainrossi, 2018-07-17:
This question has been split into several issues which will be answered separately : #4 #5 #6 #7

closed 2018-10-15 by romainrossi