romainrossi/make_your_robot_tick #9: Grading pattern

Ganeshvadhri, 2018-08-25:
1. It is mentioned that the robot that delivers the maximum amount of energy will be declared as the winner,then what is the criteria of points? 
2. How are the points awarded for the energy collected?
3. Should the robot follow the white line?

romainrossi, 2018-08-27:
1. Scores for a match takes into account the quantity of energy delivered to the Energy Storage Station (ESS for short). The ESS measures the delivered quantity of energy and display it at all time during the match. The relationship between a delivered Joule and a point will be adjusted so if the energy delivered by the fast charging station is completely delivered to the ESS, 500 points are awarded. Respectively, the accelerated station can give 250 points and the regular about 66 points. This is so the main way to score is by delivering energy to ESS and not by bonuses.
2. Points are not awarded for energy _collected_, only for the energy _delivered_ to the ESS and bonuses for various actions : leaving starting area, touching stations ... see section 5.1 page 10 of the rulebook for details.
3. White lines are here to help the robot to navigate in the game field but it is not mandatory to follow them if you want to implements another navigation strategy.

romainrossi, 2018-09-23:
It seems I made a calculation mistake in the points for accelerated and regular sources. Please see the array in issue #18 for updated information.

closed 2018-10-15 by romainrossi