romainrossi/make_your_robot_tick #13: Regarding QUALIFYING rounds and diameter of ENERGY STORAGE STATION

Ganeshvadhri, 2018-09-20:
1)The diameter of ENERGY STORAGE STATION is not mentioned anywhere in the rule book .
2)It is given in the rule book that the teams that have been approved can participate in the two qualifying rounds of the competition,also a ranking will be maintained on the basis of the points scored(the total of the two scores) to determine the teams that will qualify for the final round.
So how will the two qualifying rounds be conducted? Are the two qualifying rounds same?

romainrossi, 2018-09-21:
As your question is composed of two questions, I will split it as two "issues" and answer each of them.

romainrossi, 2018-09-21:
See answers in issues #14 and #15 .

closed 2018-10-15 by romainrossi